arduino-libraries/MadgwickAHRS #1: Madgwick Implementation

mjs513, 2016-02-16:
Hi all

I have been using these filters for the last couple of years and there are a couple of things that I am confused about with Curie implementation.  

You have the sample frequency set to 512hz but all the implementations that I have seen and what I use is that the sample frequency is calculated with the main (sketch) and passed to the filter. The second is that I would recommend that betaDef be set within the sketch to allow the user to tune the filter.

The last is that there is an error in the source code that is on Seb Madwick website in the "c" file.  Here is the reference: [http://diydrones.com/forum/topics/madgwick-imu-ahrs-and-fast-inverse-square-root?commentId=705844%3AComment%3A1755084](url)

The last thing is that gx, gy, gz should be radians/second and the way the sample is implemented it is passing these values in degrees per second - I am assuming I am reading the BMI160 data sheet and BMI160.ccp correctly.

The reason I bring this up is because if I make these changes I wind up with quite a bit of drift.

I can do a push for the changes to the library if you would like.

Thanks for your help
Mike

facchinm, 2016-02-17:
Hi Mike,
if you could open a pull request resolving the issues you have found it would be awesome!
Thank you very much

mjs513, 2016-02-19:
Hi facchinm

I have not forgotten about this - been busy trying to get rid of a very serious yaw drift as a result of making the changes.  It appears that the z axis is rather sensitive to temperature - non linear except in a certain range.  It also appears that the temperature of the BMI160 just gets rising and never really settles down - think they need a better a heat sink :) .

I also added the MahoneyAHRS to the mix as well as generating a couple of examples sketches that I will upload once I have it all sorted out.

I also had a BASH problem with the 101 that I believe you helped me with over at the Arduino forum - thanks for that.  I will push it once I have it all sorted out.

Thanks
Mike

mjs513, 2016-02-24:
Pull request submitted. I'm going to go ahead and close this issue.

closed 2016-02-24 by mjs513