arduino-libraries/MadgwickAHRS #10: Recommend Option be added to return the quaternions

mjs513, 2016-03-31:
Besides YPR the quaternions are useful to calculate dynamic acceleration as well as other coordinate transformations.  I recommend a getQ function be added to the filter files.  In MadgwickAHRS.cpp you could add:
```
void Madgwick::getQ(float * q) {
	q[0] = q0;	
	q[1] = q1;
  	q[2] = q2;
  	q[3] = q3;
}
```
while in MadgwickAHRS.h under public functions:

`	void getQ(float * q);`

In the visualizer sketch you would need to add:
`float q[4];`
and obviously to call the new function:
`filter.getQ(*q);`

R.
Mike

PaulStoffregen, 2016-04-02:
Yes, will do.

mjs513, 2016-04-02:
Thank you Paul, appreciate it.

closed 2017-03-14 by mjs513

GMcDowellJr, 2021-02-28:
> Besides YPR the quaternions are useful to calculate dynamic acceleration as well as other coordinate transformations. I recommend a getQ function be added to the filter files. In MadgwickAHRS.cpp [...]

Looks like this was never implemented. Curious if there's something wrong with it as I'm struggling to implement it myself.