arduino-libraries/MadgwickAHRS #13: Orientation visualizer understanding problem

vasanthsai, 2016-04-10:


PaulStoffregen, 2016-04-10:
Did you watch the video demo?
https://www.youtube.com/watch?v=SHJ2n1Kwe58

This test is very easy to replicate.  Just use that small program running on any Arduino board with 3 pots connected, so you can control the 3 inputs easily.  If you imagine the visualization is an airplane instead of a circuit board, you can easily tell the old code is wrong and the new code visualizes roll, pitch & yaw correctly.

It issue is a matter of extrinsic vs intrinsic rotation.  Background material can be read here:
https://en.wikipedia.org/wiki/Euler_angles
http://math.stackexchange.com/questions/1288207/extrinsic-and-intrinsic-euler-angles-to-rotation-matrix-and-back

Here's the conversation where the problem was noticed and discussed.
https://forum.pjrc.com/threads/33328-Prop-Shield-Beta-Test?p=99941&viewfull=1#post99941

Look, I don't have a lot of time to write a lot more for you, and really everything has been said far better on those many pages.  But still...

> Can you please explain the following matrix in terms simple rotateY , rotateX, rotateZ functions

Ok, really quick.... Imagine you're taking a flight within the United States, and imagine a coordinate system where the X axis is West to East, and the Y axis is South to North, as if you had a regular map of the USA on a standard X-Y grid.

I believe we can agree when your plane begins its descent to land, that's negative pitch.  Roll involves the wings tilting, and pitch involves the nose of the plane pointing upwards vs downwards.

So, if you're flying from Seattle to San Diego, this (wrong) code in the visualizer actually works fine.

      rotateX(pitch); // RotateX pitch value

But imagine instead you're flying from Seattle to New York.  As your plane pitches downward to land in NYC, the plane itself is rotating about the Y-axis (the one drawn vertically on your USA map).  This wrong code will visualize the wings tilting on your NYC descent, because it always applies pitch as a rotation about the horizontal axis.

The matrix approach correctly visualizes negative pitch as the nose of the plane pointing downward, regardless of which where the plane is actually heading (yaw).

Hopefully you can understand this.  I really can't spend more time explaining.  If you still can't see it, I highly recommend you connect any Arduino with 3 pots and run that tiny program to send roll, pitch and yaw.  If necessary, edit the code to draw an airplane instead of a circuit board.  When you do this, I'm confident you will quickly see the old code is very wrong.

vasanthsai, 2016-04-10:
thanks for the helping out !!
Really appreciate the help !!

vasanthsai, 2016-04-12:
https://github.com/PaulStoffregen/MahonyAHRS/issues/2