arduino-libraries/MadgwickAHRS #15: Pitch and yaw reversed in Visualizer

descampsa, 2016-09-01:
I  noticed a minor issue in the visualizer example : the pitch and yaw angles are reversed.
When i rotate the board in one direction along one of these axes, the visualization rotate in the opposite direction.

PaulStoffregen, 2016-09-01:
Are you using the visualizer that comes with this library, or the buggy one from Arduino's website?

https://github.com/01org/corelibs-arduino101/issues/151

descampsa, 2016-09-02:
I am using the visualizer that comes with the library, and the last git revision.

mjs513, 2017-03-14:
Hi Paul.  Know you are busy but have a really quick question.  @hanyazou (author of the BMI160 lib) gave me code for using the BMI160 FIFO.  Had trouble at first but finally got it working with his help.  So I incorporated into the visualizer code and works fairly well but having problems with orientation in the Processing visualizer.  Don't want to say exactly until I can make sure its not just me. Can you tell me where the transformation matrix came from that is in the code?  Looking at a RPY rotation matrix and trying to align it with the matrix I  have seen.

Thanks
Mike