arduino-libraries/MadgwickAHRS #17: Warnings and failure

BallscrewBob, 2017-08-23:
In the desktop IDE I get a couple of  "warnings" but in the ONLINE editor CREATE I get a compilation failure with the following output.

`
/home/admin/builder/arduino-builder/latest/Madgwick-1.2.0/src/MadgwickAHRS.cpp: In static member function ‘static float Madgwick::invSqrt(float)’:
/home/admin/builder/arduino-builder/latest/Madgwick-1.2.0/src/MadgwickAHRS.cpp:234:20: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
long i = *(long*)&y;
^
/home/admin/builder/arduino-builder/latest/Madgwick-1.2.0/src/MadgwickAHRS.cpp:236:16: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
y = *(float*)&i;
^
`

ErikBob, 2019-08-29:
C++ doesn't like Type Punning. You would have to deactive checking for strict-aliasing in your compiler, which I wouldn't recommend. Simply replace fastInvSquare() by math.h 1/sqrt(). The speedloss is insignificant.

thekunalsaini, 2020-03-20:
@facchinm can you explain how to rectify this error(approach)

ErikBob, 2020-08-18:
Simply replace fastInvSquare() by math.h 1/sqrt()

larsinka, 2022-01-26:
I am with thekunalsaini as I get the same error and don't understand the fix. Do I have to search and replace fastInvSquare() by math.h 1/sqrt()? In what file? Sorry if its a stupid question but I just don't understand it.

PaulStoffregen, 2022-01-26:
> In what file?

https://github.com/arduino-libraries/MadgwickAHRS/blob/0a909d25d0878f0c24525aeb4553cdd90473f50b/src/MadgwickAHRS.cpp#L231

larsinka, 2022-01-26:
Sorry but if I replace `float Madgwick::invSqrt(float x) { ` with `float math.h 1/sqrt(float x) {` it (understandably) throws an error.

ErikBob, 2022-01-28:
math.h is the header containing sqrt()
you can replace 
`float Madgwick::invSqrt(float x) {
	float halfx = 0.5f * x;
	float y = x;
	long i = *(long*)&y;
	i = 0x5f3759df - (i>>1);
	y = *(float*)&i;
	y = y * (1.5f - (halfx * y * y));
	y = y * (1.5f - (halfx * y * y));
	return y;
}`

by

`float Madgwick::invSqrt(float x) {
	return 1/sqrt(x);
}`