arduino-libraries/MadgwickAHRS #20: Errors calculating angles and wrong angles

RobertReichel, 2018-09-14:
Hello!

Whenever I use the code you provided, I get the following two errors:

- When using randomly chosen Quaternions, the filter won't be able to calculate the expected euler angles and give me the error that the argument in a trigonometric function (sin, cos etc.) is wrong.

- When the filter runs without an error, the euler angles I read with the getRoll etc. methods are completely wrong and when I rotate my sensor 360° around one axis, the corresponding euler angle changes only in around 50-60°

I update the filter with a frequency of 512 Hz, changed my beta to values between 1 and 0.1 and the gyroscope values are in degrees per second. Do you know a solution or what possibly might be my problem?

Kind Regards,
Robert

ErikBob, 2019-08-29:
Did you account for sensor-offsets?

MrStashley, 2020-06-02:
@ErikBob @RobertReichel 
Hi, I am having a similar problem. I never get errors, but when using the pitch yaw and roll values to control a camera in unity, I am only able to move the camera a couple of degrees no matter how much I move the sensor, and the values don't seem to correlate correctly with movements of the Arduino. I am using a frequency of 119hz, which is supposedly the update frequency of my IMU sensors and I haven't set the beta because it is not included in the constructor. I have not calibrated my Arduino sensors yet but I assumed that I would be able to achieve an approximate rotation without calibration and I assumed that because I was getting dramatic error that something else was wrong, but is it possible that calibration could be my problem?