arduino-libraries/MadgwickAHRS #33: Yaw drift LSM6D3S

vlsireddy, 2022-02-04:
Hi,

1. I see Yaw drift with the example code. using LSM6D3S with Arduino cc.
2. I looked at various forums and see that 9DoF using Magnetometer is used to compensate gyro.
3. without magnetometer, the drift could be compensated by few methods. Is this already part of this library [i could not find such code] and  if not is there any reason why it is not yet taken up.

Thanks

InputBlackBoxOutput, 2022-03-13:
Hi @vlsireddy, I am using an MPU6050 with an Arduino Uno. I also observe a drift in the yaw values while the roll and pitch values are pretty stable. Have you found a solution that will compensate for the drift without using a magnetometer?

vlsireddy, 2022-03-14:
Hi,

The experiment required accurate Yaw.
So I used a BNO055 9DoF with an Arm chip integrated sensor which does all
the Algo and gives out pure values. didn't pursue Yaw drift compensation
without magnetometer.

Thanks

InputBlackBoxOutput, 2022-03-14:
@vlsireddy No worries. I was going through the MPU6050 datasheet and found that it has a built-in motion processor that does sensor data fusion. I'll try it out when I get the time. 

Thanks!